ZuluSCSI/ZuluSCSI-firmware #565: Strange issues with Akai S3x XL types

chickeneps, 2025-05-04:
[zululog20.txt](https://github.com/user-attachments/files/20028935/zululog20.txt)

I have two clients, one with a Akai S3000XL and a Pico Slim, and the other a Akai S3200XL and a Blaster. The 3200 uses the same SCSI electronics as the 3000. And I personally have a S3000XL here.

Both report the same types of errors with their Akai's as it boots up and (tries to) recognize the ZuluSCSI's - "waiting for hard disk ready" etc. Both have images I supply that are fully formatted and good.

I've fully tested both the Blaster and Pico Slim with my S3000XL with no problems at all. And since it's two reports (at the same item) I wanted to report it. I am including the log from the S3000XL/Pico client - the other is a bit out of patience so I'm not sure if I'll get a log from him. (I'll post it here if I do.) It is certainly possible that the S3000X/Pico example's SCSI just isn't working, but them the S3200XL/Blaster example is the same report yet he has another ZuluSCSI (RP2040) from another machine and that DOES work in that S3200XL.

So here's the log, is there any knowledge that can be gleaned from it?

aperezbios, 2025-05-04:
@chickeneps can you have them try setting EnableSCSI2 to 0 in their zuluscsi.ini files?

Also, it would be a lot more helpful if we could get a debug log.

chickeneps, 2025-05-04:
See above, I did post a debug log (that was the entire purpose of the report: "So here's the log, is there any knowledge that can be gleaned from it?")

aperezbios, 2025-05-04:
it's not a debug log:
```[12ms] DIP switch settings: debug log 0, termination 1```

aperezbios, 2025-05-04:
Durr, yep it sure is, but to the first point in my initial comment, can you have them disable SCSI 2 mode and see if it makes a difference?

aperezbios, 2025-05-04:
@chickeneps you could also have the person with the Pico Slim try flashing an older version of the firmware:

https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/download/v2025.02.21/ZuluSCSI_Pico_2025-02-21.bin

chickeneps, 2025-05-04:
Client is quickly responsive (nice); he'll try both suggestions shortly tonight, and apparently he's got a S1000 and a MPC2000XL, so he'll try the ZuluSCSI Pico Slim on those. (Problem is the cabling as the S1000 uses CEN50 and the MPC2000XL uses MDB50, otherwise he would've already tried it.)

chickeneps, 2025-05-04:
Client says neither EnableSCSI2 and the older firmware affect the experience. The Akai is supposed to start right up, and when you press LOAD, you see the contents of ID 5. But his Akai errors and stalls when booting and can't even get in.

Really, though, it could be his sampler as he's never tried it with any other SCSI drive. I was just asking if the logs revealed anything that proved it's his sampler, or other. When he tries this Pico Slim with the S1000 and/or MPC2000XL, it say much more about the ZuluSCSI he has.

chickeneps, 2025-05-05:
I think we should hold off figuring this out for now, the client tried it on his S1000 and got the same error. I'm having him send it back. It's a strange thing because all Pico Slims (all ZuluSCSI, for that matter) get tested for read and write and format before they ship, and all the LED blink diagnostics succeed on his end. It's just that the samplers hate that particular unit. Same thing works great on mine.

chickeneps, 2025-05-08:
I got the ZuluSCSI Pico Slim back from the customer, worked 100% here. I have no idea what his issue was, we covered everything. That was with his S3000XLand S1000. He elected for it to be sent back to him, and he's going to try harder to get it to work on his S1000 and MPC2000XL. We should close this ticket.

closed 2025-05-08 by chickeneps