ZuluSCSI/ZuluSCSI-firmware #572: All Writes Fail: Watchdog Timeout

mookiedog, 2025-05-13:
I am using a Zulu RP2350B Blaster in an HP logic analyzer, model 16702A.

My SD cards are formatted in exFat, and contain 2 files:

- CD3.iso
- HD6.img

where CD3.iso contains the iso image of the hard disk creation CD, and HD6 is a 6 gigabyte blank hard disk image.

The analyzer sees both SCSI devices when it starts up, as expected.

The analyzer is set up to boot from CD3. I can see that this is working because the system loads the installation software from the CD ISO, which starts running as expected.  After some keyboard inputs to the analyzer installation software, it begins the installation, printing one last message that it is installing the software.

From that point on, the activity LED on the Blaster comes on solid for about 14 seconds, then goes off for 2 seconds, then on solid for 14, off for 2, forever.

I enabled logging via the dipswitch on the blaster board.  As the analyzer boots, I can see SCSI traffic showing the analyzer talking to the HD, then the CD. Once I tell the analyzer to boot from the CD, the log shows tons of data reads from CD3. The log shows that at the very end, it finally starts talking to HD6 again. A number of status-like commands appear to be processed by HD6 as expected, but as soon as the bus log shows an attempt to write data to HD6, it gets a "watchdog timeout" after 14.3 seconds. Two seconds later, the write gets retried. 14.3 seconds later, another watchdog timeout. This sequence repeats forever. No write ever succeeds.

I have tried 3 different SD cards - all act in identical fashion.

Here is a shortened version of the debug log zululog.txt. I stripped out 99% of the read operations from the CD3 device to highlight the startup activity and then the write failures at the end.

[zululog-1.txt](https://github.com/user-attachments/files/20195481/zululog-1.txt)

mookiedog, 2025-05-13:
I forgot to mention that I also tried updating the Blaster to the latest firmware, 2025-05-12. Same watchdog timeout.

PetteriAimonen, 2025-05-14:
It looks like a few ack/req pulses are getting lost somewhere due to some kind of signal quality issue.

Can you try with this in zuluscsi.ini:

```
[SCSI]
MaxSyncSpeed = 0
```

PetteriAimonen, 2025-05-14:
Alternatively, you can try this firmware build that fixes the SCSI-5 synchronous mode timing on Blaster:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/15016308222

mookiedog, 2025-05-14:
Signal quality on the logic analyzer's SCSI bus, or on the ZuluSCSI board?

I tried the zuluscsi.ini change first, and it works! Thank you.

I will also try the new firmware. Do you have a recommendation for which fix to go with if they both work?

PetteriAimonen, 2025-05-14:
I initially suspected signal quality between the host SCSI and ZuluSCSI. That could have been caused by cable or by either end.

But I now think that was incorrect, and the real reason was the Blaster board having a too short pulse (about 50 ns) on REQ line.
SCSI specification is minimum 30 ns pulse for synchronous speeds faster than 5 MB/s, but for 5 MB/s the spec requires minimum 90 ns.
What caused this is the higher clock speed of Blaster, which led to limitation in the delay time specification.
That is now fixed in the new firmware, by dividing the clock by two when using the 5 MB/s mode.

If both work, synchronous mode (new firmware with default configuration) will give slightly better performance.
But the performance difference is likely small, maybe 20% or so.

aperezbios, 2025-05-14:
@mookiedog the fix @PetteriAimonen described has now been incorporated into the [v2025.05.14](https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/tag/v2025.05.14) release, which I just cut, so please test that instead of the above-referenced URL that @PetteriAimonen provided

mookiedog, 2025-05-14:
I have tested the v2025.05.14 release and can confirm that it has also fixed the issue. Thank you for the very timely service!

Separately, I would like to say that your Blaster board works great in my analyzer! Being able to emulate multiple SCSI devices at the same time allowed me to create a fake CD-ROM drive holding the analyzer's recovery ISO image as well as a blank fake hard disk image. The analyzer installed from fake CD-ROM to fake disk resulting in a working system again. Cheers!

aperezbios, 2025-05-14:
Confirmed fixed [v2025.05.14](https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/tag/v2025.05.14) firmware release

closed 2025-05-14 by aperezbios