ZuluSCSI/ZuluSCSI-firmware #583: CD image does not work in DOS

darkalpha84, 2025-06-07:
Greetings,
i received my ZuluSCSI Blaster with CD Audio extension.

Today i could test it and was especially curious for the CD Audio.

Story short: In Windows 2K the cd image in mixed mode (data track and audio tracks) works.
In DOS i can't access the disc with the error:
[43201ms] WARNING: Host attempted CD read at sector 0x001FFF7A+0x00000001, exceeding image size 0x000000000EB37730

I'm using the original Tekram DC 390 DOS drivers, the host adapter and the device gets detected (aspi driver). The tekram scsi cd driver detects the drive. shscdx (mscdex replacement from free dos) can find the drive and assigns a driveletter.

Toolbox enabled.

Any clues? Thank you :)

morio, 2025-06-10:
@darkalpha84 could you provide the full log with debug mode on? Flip the middle switch the on the DIP switch to "ON" or add `debug=1` under `[SCSI]` in the `zuluscsi.ini` file.  Also could you provide a copy of the ".cue" file?
Thanks!

darkalpha84, 2025-06-11:
Thanks for your reply. Attaching requested data. Cue file as .txt because .cue is not allowed.

[zululog.txt](https://github.com/user-attachments/files/20695138/zululog.txt)

[DESCENT_II.txt](https://github.com/user-attachments/files/20695144/DESCENT_II.txt)

morio, 2025-06-11:
@darkalpha84 

With Windows 2k was the Blaster also hooked up via the Tekram 390?

By any chance could you provide a link to the bin file?

The following are notes are more for me:

There is definitely something strange going on
```
[104096ms] DBG -- BUS_FREE
[104097ms] DBG ---- SELECTION: 2
[104097ms] DBG ---- MESSAGE_OUT
[104097ms] DBG ------ OUT: 0xC0 
[104098ms] DBG ---- COMMAND: Read6
[104098ms] DBG ------ OUT: 0x08 0xFF 0xFF 0x7A 0x01 0x00 
[104099ms] DBG ------ Read CD: 1 sectors starting at 2097018, track number 9, track mode 0, sector size 2352, main channel 0x10, sub channel 0x00, data offset in file 637219040
[104100ms] WARNING: Host attempted CD read at sector 0x001FFF7A+0x00000001, exceeding image size 0x000000002B747290
[104102ms] DBG ---- STATUS: 2 CHECK_CONDITION, sense 0x00002100
[104106ms] DBG ---- MESSAGE_IN
[104106ms] DBG ------ IN: 0x00 
[104106ms] DBG -- BUS_FREE

```

The read 6 command `OUT: 0x08 0xFF 0xFF 0x7A 0x01 0x00 ` is accessing LUN 7 (top 3 bits of the second byte), LUNs aren't supported and don't make sense in this case. Neither does the starting LBA of 0x1FFFF7A which would be a about 1GB offset (512 bytes per sector) and as a CD-ROM drive about 4GB (2048 to 2532  bytes per sector). It does a READ(10) earlier in the log which looks fine
```
[77521ms] DBG -- BUS_FREE
[77521ms] DBG ---- SELECTION: 2
[77522ms] DBG ---- MESSAGE_OUT
[77522ms] DBG ------ OUT: 0xC0 
[77522ms] DBG ---- COMMAND: Read10
[77523ms] DBG ------ OUT: 0x28 0x00 0x00 0x00 0x00 0x11 
[77523ms] DBG ------ OUT: 0x00 0x00 0x01 0x00 
[77524ms] DBG ------ Read CD: 1 sectors starting at 17, track number 1, track mode 3, sector size 2352, main channel 0x10, sub channel 0x00, data offset in file 39984
[77526ms] DBG ---- DATA_IN
[77526ms] DBG ---- Unaligned access to image, falling back to SdFat access mode
[77528ms] DBG ---- Total IN: 2048 OUT: 0 CHECKSUM: 6890
[77528ms] DBG ---- STATUS: 0 GOOD
[77532ms] DBG ---- MESSAGE_IN
[77532ms] DBG ------ IN: 0x00 
[77532ms] DBG -- BUS_FREE

```

darkalpha84, 2025-06-12:
DOS 7.1 and Win2K are running on the same PC via bootloader.
Win2K using the official Tekram driver.
In Win2K everything works fine. Can access the data partition and use the cd player software provided by Win2K to play the cd tracks.

darkalpha84, 2025-06-14:
I can't provide a link to an image, because its a licensed video game, even though its really old now.

morio, 2025-06-16:
@darkalpha84 
Could you try this firmware by either putting the `.bin` file on your SD card and powering the board or holding down bootloader button, powering on the board, and copying the `.uf2` file to the `RP2350` USB drive?
[ZuluSCSI_Blaster.zip](https://github.com/user-attachments/files/20761949/ZuluSCSI_Blaster.zip)

Then boot your DOS partition and capture the `zululog.txt` file with debug on.

After that, add create a file called `zuluscsi.ini` on the root directory of the SD card and add this text to the file.
```
[SCSI]
ForceTOC = 1
```
Boot your DOS partition and capture that `zululog.txt` with debug on and post both `zululog.txt` files here

Thanks!

darkalpha84, 2025-06-17:
Thank you for the test firmware!
This firmware + [SCSI] ForceTOC = 1 did the trick!

I will append zululogA.txt    (new firmware + ForceTOC = 0)    and zululogB.txt (new firmware + ForceTOC = 1)
I will test some more images in the next days and report back here even if they work - so you know you made it work :).

[zululogA.txt](https://github.com/user-attachments/files/20780588/zululogA.txt)

[zululogB.txt](https://github.com/user-attachments/files/20780589/zululogB.txt)

morio, 2025-06-17:
@darkalpha84 could you please test this firmware: https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/15718417863
And remove the 
```
[SCSI] 
ForceTOC = 1
```
from `zuluscsi.ini`
and turn debug off.

It should just work with no special settings
Thanks!

darkalpha84, 2025-06-19:
I can confirm, this firmware works with [SCSI] 
ForceTOC = 0.

Thanks for fixing!
Now i would need help to convert my cd images from mode 2 to mode 1, because your firmware only supports type 1.

morio, 2025-06-27:
I think this is a problem:
```
[81495ms] DBG ---- COMMAND: CDROM Read SubChannel
[81496ms] DBG ------ OUT: 0x42 0x02 0x40 0x00 0x00 0x00 
[81496ms] DBG ------ OUT: 0x00 0x00 0x10 0x00 
[81497ms] DBG ---- Unsupported subchannel request
[81497ms] DBG ---- STATUS: 2 CHECK_CONDITION, sense 0x00002400
[81503ms] DBG ---- MESSAGE_IN
[81503ms] DBG ------ IN: 0x00 
```
I'll see if I can implement the request type.

aperezbios, 2025-06-27:
@darkalpha84 see above, can you test this as time permits? Thanks!