ZuluSCSI/ZuluSCSI-firmware #587: IBM SCSI-2 Fast/Wide Adapter/A for PS/2 Compatibility Issue

hlub69, 2025-06-12:
I have been trying to get the ZuluSCSI Blaster and ZuluSCSI Compact to work with the IBM SCSI-2 Fast/Wide Adapter/A (Corvette) on an IBM PS/2 Model 77.  With default settings and no zuluscsi.ini file on the microSD card, the IBM adapter recognizes the Zulu and all the emulated devices.  However, it cannot write to any of them.  This is true for both the Blaster and the Compact.  When I connect either Zulu to the on-board SCSI (I believe it is an IBM Spock adapter), both Blaster and Compact work fine with read and write operations.

Now here's the weird part.  Back to the IBM Corvette card and only with the Blaster, I placed the microSD into an SD Adapter and if I switch the SD Lock ON, I can read and write to the Blaster.  However, since the Compact uses a microSD and there is no Lock/Unlock on microSD cards, it's a no-go.

In summary, IBM SCSI-2 Fast/Wide Adapter/A (Corvette) reads and write when SD card is set to LOCKED position and only read when SD card is set to the UNLOCKED position.

aperezbios, 2025-06-12:
@hlub69 Hmm, that's very bizarre, and on the surface makes no sense. I have a PS/2 486SLC and do not have this problem with a ZuluSCSI Compact. Which firmware version are you using on both boards? 

Can you please enable debug mode on the Blaster via either the DIP switch or via the ini file, and then try this again with your ZuluSCSI Blaster board, with the SD card lock switch in the correct/on position? It will help us understand what's happening when the writes are attempted. Once you have the debug log, please attach it here, and feel free to compress it if it's large enough to warrant it.

aperezbios, 2025-06-12:
@hlub69 also, it is electrically impossible for the state of the write protect switch on a full-size SD card to be read by the firmware, because that pin on the SD card socket is not connected to anything on the microcontroller. This is the relevant portion of the ZuluSCSI Blaster schematic, where you can see that the WRITE_PROTECT pin on the SD card socket is not routed anywhere.

[image]

hlub69, 2025-06-12:
[zululog_writeprotectoff.txt](https://github.com/user-attachments/files/20717352/zululog_writeprotectoff.txt)

I observed something.  On a cold boot, when the SD card is in the UNLOCK position, it won't write.  Without turning off the computer, I removed the SD card and put it in the LOCK position, the computer is able to write to the emulated drive.  Now, I turn off the computer and don't touch the SD card, then turn the computer back on (SD card still in the LOCK position), it no longer write.  Pop out the SD card and move the tab to the UNLOCK position and pop it back in, it writes.

I tried this on 2 different cards, SanDisk Ultra 32GB and a Samsung Pro 32.  Both behave exactly the same.

hlub69, 2025-06-12:
Here's my zuluscsi.ini.

[zuluscsi.ini.txt](https://github.com/user-attachments/files/20717449/zuluscsi.ini.txt)

aperezbios, 2025-06-12:
@hlub69 Please try without SpeedGrade set to TurboMax, and without MaxSyncSpeed=20 set. There's nothing obvious in the error log that would indicate that there's a write issue.

Also, all of the other parameters you have configured in the INI file are the default values, so they effectively change nothing, and aren't necessary at all.

Finally, is there a reason why you have MaxSyncSpeed set to 20 on a what appears to be a system that doesn't support 20MB/sec SCSI? 

```
SelectionDelay = 255
Quirks = 0
EnableUnitAttention = 0
EnableSCSI2 = 1
EnableSelLatch = 0
EnableParity = 1
MapLunsToIDs = 0
SpeedGrade = TurboMax
MaxSyncSpeed = 20
```