ZuluSCSI/ZuluSCSI-firmware #589: dfu-util fails to flash

ltning, 2025-06-16:
Hi! Wanted to use the Azuluscsi 1.1 with my 486 and Adaptec 1542 scsi controller. 
Used a microsd card of 32GB with a single exfat partition, put the default zuluscsi.ini on it.  
Was not detected by the SCSI boot rom, so I wanted to try a firmware upgrade. 

Followed the instructions for ZuluSCSI 1.1 at https://github.com/ZuluSCSI/ZuluSCSI-firmware?tab=readme-ov-file#programming--bootloader

Tried flashing several times, also with lower transmit chunk sizes and different USB ports and cables. Tested on Ubuntu 25.04 and FreeBSD 14.3, different machines.

Not sure what I'm missing? Seeing this on both of my AzulSCSI boards.

It finds the device just fine:
```
$ sudo dfu-util -l
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found DFU: [28e9:0189] ver=1000, devnum=31, cfg=1, intf=0, path="3-4", alt=1, name="@Option Bytes  /0x1FFFF800/01*016 g", serial="5VCC"
Found DFU: [28e9:0189] ver=1000, devnum=31, cfg=1, intf=0, path="3-4", alt=0, name="@Internal Flash  /0x08000000/512*002Kg", serial="5VCC"
```

Next, tried to flash (the firmware corrupt message is because this was not my first attempt):
```
$ dfu-util --alt 0 --dfuse-address 0x08000000 --download ZuluSCSIv1_1_plus_2025-06-11.bin 
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release
Opening DFU capable USB device...
Device ID 28e9:0189
Device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(10) = dfuERROR, status(10) = Device's firmware is corrupt. It cannot return to run-time (non-DFU) operations
Clearing status
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading element to address = 0x08000000, size = 163084
Erase   	[=====                    ]  20%        32768 bytesdfu-util: dfuse_download: libusb_control_transfer returned -9 (LIBUSB_ERROR_PIPE)
dfu-util: Error during special command "ERASE_PAGE" download
```

Testing on a different machine (FreeBSD 14.3), similar issue. The LIBUSB error is different but that's likely a porting artifact (different OS):

```
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

libusb version 1.0.0 (2016)
dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release
Opening DFU capable USB device...
Device ID 28e9:0189
Device DFU version 011a
DFU attributes: (0x0b) bitCanDnload bitCanUpload bitWillDetach
Detach timeout 255 ms
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(5) = dfuDNLOAD-IDLE, status(0) = No error condition is present
Aborting previous incomplete transfer
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Memory segment at 0x08000000 512 x 2048 = 1048576 (rew)
Downloading element to address = 0x08000000, size = 163084
Erasing page size 2048 at address 0x08000000, page starting at 0x08000000
   Poll timeout 100 ms
   Poll timeout 0 ms
Erasing page size 2048 at address 0x08000800, page starting at 0x08000800
   Poll timeout 100 ms
   Poll timeout 0 ms
Erasing page size 2048 at address 0x08001000, page starting at 0x08001000
   Poll timeout 100 ms
   Poll timeout 0 ms
Erasing page size 2048 at address 0x08001800, page starting at 0x08001800
   Poll timeout 100 ms
   Poll timeout 0 ms
Erasing page size 2048 at address 0x08002000, page starting at 0x08002000
   Poll timeout 100 ms
   Poll timeout 0 ms
Erasing page size 2048 at address 0x08002800, page starting at 0x08002800
dfu-util: dfuse_download: libusb_control_transfer returned -4 (LIBUSB_ERROR_NO_DEVICE)
dfu-util: Error during special command "ERASE_PAGE" download
```

aperezbios, 2025-06-16:
@ltning unless your ZuluSCSI board does not have firmware on it already, you should not be using DFU as the update mechanism. Please provide a copy of the zululog file so we can determine which firmware version you're currently running.

aperezbios, 2025-06-16:
@ltning the log file is written directly to the fat-formatted SD card, regardless of whether or not it's detected as a SCSI device. That's completely irrelevant here. 

What does the activity LED do when you power on the board with an SD card inserted? How many times does the LED blink before repeating its pattern?

ltning, 2025-06-16:
There is no blinking, with or without SD card, since the firmware is now corrupt. I should probably have tried with fat32 first, it could be that the firmware originally was too old to handle exfat?