ZuluSCSI/ZuluSCSI-firmware #594: E-mu Emax and ZULUSCSI drive

soundsubs, 2025-06-18:
Trying to bring an old E-mu Emax into this century with ZULUSCSI: 
I acquired yet another Baldwin Emax in mint condition, and I'm trying to upgrade it with Gotek Floppy and ZuluSCSI all in one: 
[Gotek-SCSI – SamplerZone.com](https://samplerzone.com/products/gotek-scsi)
I eventually got the Gotek to work after calling SamplerZone, but I can't get anything out of the ZuluSCSI like all 5(?) Emaxes I've done before with SCSI2SD... 
Of course I'm using the Emax Plus OS 1.0 as I've done multiple times, and see SCSI ID = 0 with the Transpose button. So that part is alive. 
It is powered via a splitter cable (for the Gotek and the Zulu) and has a blinking LED to indicate its getting power. 
It is blinking 5 times in a row. Not sure what that means? 
However, I'm getting nothing but "HD Catalog Error" and "Format Error" when trying to access anything on the HD. When scanning, it just sits there forever.  

What I've tried: 

- First, I made sure the board is PC333 (version 3) and has the additional traces on the underside
- Then I used one of my AMD SCSI chips-- I've tried 2 different ones and I have more on order
- Changed the 50 pin SCSI cable out, with two new ones. 
- Changed the orientation of the 50 pin SCSI cable multiple times.
- Changed the formatting of the SD card about 20 times, including no format, with banks via EMXP, etc. 
- Today I changed out the PAL chip with a NOS one. No dice. 

Another friend indicated I might need to upgrade the ZuluSCSI firmware, especially for the Emax. 

Any ideas here?

-Shane

aperezbios, 2025-06-18:
@soundsubs please provide us with a zululog.txt file. We don't particularly need a ton of information about the sampler itself. Right now we're flying blind because we have no idea how you have your ZuluSCSi configured. The zululog.txt file is written to the SD card by the firmware every time it boots.

chickeneps, 2025-08-14:
This is been addressed fully and privately with our customer.

closed 2025-08-14 by aperezbios