ZuluSCSI/ZuluSCSI-firmware #600: ZuluSCSI Blaster not Detecting RM2 Module

techfixpros, 2025-06-27:
ZuluSCSI Blaster Rev. 2025e w/RM2 Rev. 2025a
Firmware 2025.06.01 - 2025.06.11
Macintosh SE/30
- Reloaded v5 Logic board by Caymac Vintage
- Re-capped analog board by me
- BabyFace Universal PSU
SanDisk Extreme 64GB u3 A2
- Apple System 7.5.3
- DaynaPort SCSI/Link driver installed
- OpenTransport 1.1.2 configured (& working when SD installed into ZuluSCSI SLIM)
- NE6.img (0k file for WiFi per Wiki)

The Blaster powered over USB, 50-pin SCSI cable (internal), or Berg connector does not make the LED on the RM2 module light up. When installed onto the Blaster, I measure ~3.3v from any of the 8 GND pins to either of the two 3v3 pins. Per the log file, "RM2 not found, upclocking", and "Wi-Fi connection failed: -4" are the error messages. The Blaster will also hang during booting Mac System 7 while the NE6.img file is on the SD card. Removing the .img allows the Blaster to fully boot up and provide a working experience, sans Wi-Fi.

I have tested the SD card and configuration by moving it back to my ZuluSCSI Slim and booting from that. NE6.img is also back on the SD card to enable Wi-Fi to work, which it does flawlessly. Tested Wi-Fi via FTP transfer, web browsing to 68kmla news, and BBSing.

Here is the log file while installed in the Blaster, and NE6.img on the SD:
`[10ms] Platform: ZuluSCSI Blaster

[10ms] FW Version: 25.06.01-release Jun  1 2025 03:51:40

[11ms] DIP switch settings: debug log 0, termination 1

[12ms] SCSI termination is enabled

[12ms] ===========================================================

[13ms]  Powered by Raspberry Pi

[13ms]             Raspberry Pi is a trademark of Raspberry Pi Ltd

[14ms] ===========================================================

[15ms] Flash chip size: 16384 kB

[15ms] SCSI target/disk mode selected by DIP switch, acting as a SCSI disk

[15ms] Reclocking with these settings are compatible with CD audio playback

[16ms] Initial Clock set to 150000000Hz

[17ms] Reclocking the MCU to 155250000Hz

[18ms] After reclocking, system reports clock set to 155250000Hz

[369ms] RM2 not found, upclocking

[369ms] Reclocking with these settings are compatible with CD audio playback

[370ms] Initial Clock set to 155250000Hz

[370ms] Reclocking the MCU to 200400000Hz

[371ms] After reclocking, system reports clock set to 200400000Hz

[371ms] Starting Core1 for audio

[414ms] SD MID: 0x03, OID: 0x53 0x44

[414ms] SD Name: SN64G

[414ms] SD Date: 6/2022

[414ms] SD Serial: 0x87C84E05

[421ms] Reading configuration from zuluscsi.ini

[427ms] Active configuration (using system preset "Mac"):

[428ms] -- SelectionDelay = 255

[428ms] -- EnableUnitAttention = No

[428ms] -- EnableSCSI2 = Yes

[429ms] -- EnableSelLatch = No

[429ms] -- MapLunsToIDs = No

[429ms] -- EnableParity = Yes

[485ms] Finding images in directory /:

[487ms] -- Using custom block size, 512 from filename: HD10_512  753 System.hda

[490ms] -- Opening /HD10_512  753 System.hda for id:1 lun:0

[491ms] ---- Configuring as disk drive drive

[491ms] ---- Drive geometry from image size: SectorsPerTrack=24 HeadsPerCylinder=16 total sectors 4096128 (divisible)

[493ms] ---- Read prefetch enabled: 8192 bytes

[493ms] -- Using custom block size, 512 from filename: HD20_512  753 Programs.hda

[496ms] -- Opening /HD20_512  753 Programs.hda for id:2 lun:0

[497ms] ---- Configuring as disk drive drive

[498ms] ---- Drive geometry from image size: SectorsPerTrack=8 HeadsPerCylinder=16 total sectors 2048128 (divisible)

[499ms] ---- Read prefetch enabled: 8192 bytes

[506ms] -- Opening /NE6.img for id:6 lun:0

[508ms] ---- Configuring as network based on image name

[511ms] -- Platform supports ROM drive up to 15612 kB

[511ms] ---- ROM drive image not detected

[512ms] SCSI ID: 1, BlockSize: 512, Type: 0, Quirks: 1, Size: 2048064kB

[512ms] SCSI ID: 2, BlockSize: 512, Type: 0, Quirks: 1, Size: 1024064kB

[513ms] SCSI ID: 6, Type: 6, Quirks: 1

[1204ms]  

[1204ms] === Network Initialization ===

[1555ms] Wi-Fi MAC: 00:80:19:C0:FF:EE

[1555ms] Connecting to Wi-Fi SSID "<REDACTED>" with WPA/WPA2 PSK

[1556ms] Wi-Fi connection failed: -4`

techfixpros, 2025-06-30:
Update: Received a BlueSCSI v2 this weekend, and now have a better sample size (ZuluSCSI Pico Slim, ZuluSCSI Blaster, and BlueSCSI v2 Desktop).

The Macintosh SE/30 does indeed provide suitable termination power to both internal and external connectors (ZS Pico and BS v2 tested fine). 

WiFi configuration and HDD images on my SD card work fine in both the v2 and Pico (am using the same card,l and images, the only difference is now there is both a zuluscsi.ini and bluescsi.ini on the card. Moving the card between the three emulators and the problem only exists when the Blaster is attempted to use.

I cannot find any youtube or forum reference of anyone else using the Blaster, nor of thr Blasters WiFi functionality, so am unsure if it's even supported/fully implemented in code.

aperezbios, 2025-06-30:
@techfixpros since this is a product support issue, we've responded to it via e-mail, and offered to send you a replacement unit.

closed 2025-06-30 by aperezbios