ZuluSCSI/ZuluSCSI-firmware #603: Toolbox API version difficult to get - variable length strings

nielsmh, 2025-06-28:
In `inquiry.c` the vendor info part of the vital product data page 0 is filled in this section:

https://github.com/ZuluSCSI/ZuluSCSI-firmware/blob/da079f16ef3c44afd2448fd3b6901fbd8a6e6b4b/lib/SCSI2SD/src/firmware/inquiry.c#L276-L281

The `INQUIRY_NAME` is defined in `ZuluSCSI_config.h`:

https://github.com/ZuluSCSI/ZuluSCSI-firmware/blob/da079f16ef3c44afd2448fd3b6901fbd8a6e6b4b/src/ZuluSCSI_config.h#L29-L38

I'm interested in extracting the TOOLBOX_API value from the data supplied from the inquiry.

As far as I'm aware, the SCSI-2 standard allocates 20 bytes for the vendor information in the vital product data. When searching for `PLATFORM_NAME`, there seem to be several definitions, selected based on the build target:

* `"ZuluSCSI Pico DaynaPORT"`
* `"ZuluSCSI Pico"`
* `"ZuluSCSI mini v1.0"`
* `"ZuluSCSI v1.0"`
* `"ZuluSCSI v1.4"`

These range from 13 to 23 bytes in length. 

Appended to this is a dash and the firmware version, which is 12 bytes in the current development state of the code. Total, the `INQUIRY_NAME` appears to be never shorter than 26 bytes, but may be longer depending on the platform and build.

This seems to be breaking the barrier for the field it is filled in to, as well as making it difficult to detect where the TOOLBOX_API version value is located. I'm considering checking for a prefix of `"ZuluSCSI`", then skipping further bytes until the first byte smaller than 0x20, and assume that will be the TOOLBOX_API version.

morio, 2025-07-01:
@nielsmh will this make things easier?
https://github.com/ZuluSCSI/ZuluSCSI-firmware/tree/fix/inquiry-name-for-toolbox

nielsmh, 2025-07-01:
Yes that would definitely make it predictable. The extended hardware and firmware revision information might be more suitable to put in the vendor specific data, starting at offset 96.

If the Vendor specific information string is limited to `"ZuluSCSI\0"` for Toolbox API version 0, I think it'll also be reasonable to expect that if the byte following `"ZuluSCSI"` is >= 0x20 then it's definitely API version 0.

A different proposal is to change the response to mode page 0x31 whenever a new Toolbox API version (that breaks compatibility) is released, such that applications that match either a prefix or full contents of the mode page, would detect non-support. In that case, it wouldn't strictly be necessary to change the inquiry data. (But there is an argument for shortening the inquiry vendor specific information string regardless, in case there are ever any plans to support SCSI-3 or later, since the reserved bytes occupied by the string currently do get defined in later SCSI versions.)

aperezbios, 2025-07-10:
@nielsmh this now implemented in the v2025.07.09 release

closed 2025-07-10 by aperezbios