ZuluSCSI/ZuluSCSI-firmware #647: 2025.08.26 seems to break SCSIKnife (Pico based)

vom513, 2025-08-27:
Hello,

Obviously the SK is a totally separate project and I would never expect this project to support that.  However since it uses the same firmware I figured there might be some value in letting you know about this.

I have an SK SCA (I think this is just the "standard" SK with a 50-SCA adapter ?).  I have this in my Sun SS20.  It's the only device on that bus, termination on.  Seems to work well.  This was with **2025.08.19** being the latest firmware I tried.

I put **2025.08.26** on the other day and got some very strange behavior.  The activity LED would go for about 5 seconds every 10-15 seconds.  This isn't the "no SD card present" pattern, this was something else.  When booting the machine, I got parity errors in the kernel (NetBSD) as well as it not even being able to mount the root FS.  I reverted to **2025.08.19** and all seems well again.

Total stab in the dark, but the one thing that caught my eye in the release notes was:

```
Standardize board system clock frequencies by @morio in https://github.com/ZuluSCSI/ZuluSCSI-firmware/pull/638
```

If you want debug logs from this let me know.  There didn't seem to be anything "bad" in the standard non-debug log with a quick glance.

Thanks.

vom513, 2025-08-27:
I captured starting with it running 08.19, and you can see the upgrade happen.  Def. seems to be some nastiness after 08.26 is running.  Let me know if there's anything else you need me to run or capture.  Thanks.

[sk-zululog-2025-08-26.log](https://github.com/user-attachments/files/22009690/sk-zululog-2025-08-26.log)

aperezbios, 2025-08-27:
@vom513 can you provide us with some additional information about the exact make and model of SD card you're using, as well as whether or not you're using the full-sized SD card or microSD card socket on the SCSIKnife?

We just briefly tried with the latest v2025.08.26 release on a SCSIKnife we have here, and were unable to immediate reproduce the SDIO checksum errors that show up 13 milliseconds from the moment the firmware initializes in the log you attached.

aperezbios, 2025-08-27:
> I can try using the native micro slot on the SK if you think there's value in that (or if that's what you did so we are testing like for like).

Yes, I think you should definitely at least _try_ using the microSD card socket, and report back if you see any difference in behavior.  That's what we tested with. 

Right now, what we need is as much hard data from you as possible, since we're not able to easily replicate the issue you've encountered.

morio, 2025-08-27:
@vom513 try these settings. The file contains zuluscsi.ini and zuluscsi_timings.ini that should be put in the root directory of the SD card.
[slower-sd-clock.zip](https://github.com/user-attachments/files/22010994/slower-sd-clock.zip)

vom513, 2025-08-27:
Yeah I think I'm back to square one.  Nothing "bad" is logged when NetBSD fails to mount root.  The command warning and speed negotiations are always something I get (last few lines).

[image]

[sk-zululog-08.26-micro-socket.txt](https://github.com/user-attachments/files/22012796/sk-zululog-08.26-micro-socket.txt)