ZuluSCSI/ZuluSCSI-firmware #650: Hangs/timeouts on Amiga SCSI-Kit IV system, ZuluSCSI 25.08.26-release

screwtop, 2025-09-02:
I'm seeing sporadic problems with my SCSIknife (a ZuluSCSI derivative from Jurassic Computing, https://github.com/Siliconinsid/SCSIknife), which is running the ZuluSCSI firmware (I updated it to version 25.08.26-release). It works for a while, then hangs, sometimes with AmigaDOS reporting I/O errors, but more often things just hang (mouse pointer still works but most UIs are nonresponsive). In the ZuluSCSI debug log I see "WATCHDOG TIMEOUT" and "scsi_accel_rp2040_finishRead timeout" errors, and the yellow drive activity light remains on for long periods. Problems persist even after rebooting the system, and I have to power down both the Amiga and the SCSIknife to get it to reset properly. Replugging the SD card will usually help if the Amiga sees no devices on the bus even after rescanning using UnitControl2.

I have occasionally seen very good performance numbers, with read speeds over 7 MB/s on the Amiga. However, it just isn't behaving reliably for some reason. I've successfully used other SCSI devices on this system (external hard drive, scanners, Zip100 drive, and a CD-RW drive).

## System Details

Hardware:
 - SCSIknife
 - SD card: SanDisk Ultra microSDHC UHS-I card, 32 GiB
 - Amiga 1200
 - Blizzard 1260 accelerator (50 MHz 68060)
 - SCSI-Kit IV host adapter (FAS216)
 - DB25 to CN50 cable, < 1 m (I've tried a couple of different ones)
 - Igloo drive enclosure from the 1990s
 - Active CN50 terminator (also tried the SCSIknife's internal termination)

Software:
 - ZuluSCSI 25.08.26-release
 - Amiga Kickstart V40.68
 - Workbench V39.29
 - `1230scsi.device` V8.5
 
The SCSIknife is the only SCSI device attached (I did also do some testing with a Zip100 drive terminating the bus). It's in an external enclosure with CN50 connectors and an internal 50-pin ribbon cable. I'm powering the SCSIknife via USB in case the host's termination power supply is inadequate.

## ZuluSCSI Logs

Here's the initial startup log from the ZuluSCSI:

```
[12ms] FW Version: 25.08.26-release Aug 26 2025 19:19:47
[12ms] DIP switch settings: debug log 1, termination 1
[12ms] SCSI termination is enabled
[12ms] =========================================================================
[12ms]  Powered by Raspberry Pi
[12ms]             Raspberry Pi is a trademark of Raspberry Pi Ltd
[12ms] 
[12ms]  ZuluSCSI RPi platform support is developed by Rabbit Hole Computing
[12ms]  and provided to you under GNU General Public License version 3.
[12ms] =========================================================================
[13ms] Flash chip size: 2048 kB
[13ms] SCSI target/disk mode selected by DIP switch, acting as a SCSI disk
[13ms] DBG Starting Core1 dispatcher
[13ms] Reclock RP2040 & Pico 1/1W based boards to standardized speed
[13ms] Initial Clock set to 133000000Hz
[13ms] Reclocking the MCU to 203200000Hz
[13ms] Setting the SDIO clock to 41MHz
[13ms] After reclocking, system reports clock set to 203200000Hz
[30ms] SD card detected, FAT32 volume size: 30424 MB
[30ms] SD MID: 0x03, OID: 0x53 0x44
[30ms] SD Name: SD32G
[30ms] SD Date: 12/2024
[30ms] SD Serial: 0xC5F949F8
[546ms] Speed grade set to Default, skipping reclocking
[586ms] Reading configuration from zuluscsi.ini
[1102ms] Active configuration:
[1102ms] -- SelectionDelay = 255
[1102ms] -- EnableUnitAttention = No
[1102ms] -- EnableSCSI2 = Yes
[1102ms] -- EnableSelLatch = No
[1102ms] -- MapLunsToIDs = No
[1102ms] -- EnableParity = Yes
[3704ms] Finding images in directory /:
[3995ms] -- Opening /HD1_OpenRetro_MacOS753.img for id:1 lun:0
[4037ms] DBG ---- Image file is contiguous, SD card sectors 3322816 to 4346815
[4037ms] ---- Configuring as disk drive drive
[4037ms] ---- Drive geometry from defaults: SectorsPerTrack=63 HeadsPerCylinder=255 total sectors 1024000 (not divisible)
[4037ms] ---- Read prefetch enabled: 8192 bytes
[4327ms] -- Opening /HD3_Amiga_Zip.img for id:3 lun:0
[4336ms] DBG ---- Image file is contiguous, SD card sectors 417856 to 614463
[4336ms] ---- Configuring as disk drive drive
[... 255 lines cut ...]
[735961ms] SyncOffset: 0 SyncPeriod 0
[735961ms] PIO Parity SM: tx_fifo 0, rx_fifo 0, pc 6, instr 0x000060C8
[735961ms] PIO Data SM: tx_fifo 0, rx_fifo 4, pc 19, instr 0x00005056
[735961ms] PIO Sync SM: tx_fifo 0, rx_fifo 0, pc 0, instr 0x00000000
[735961ms] DMA CH A: ctrl: 0x01020021 count: 0x00000006
[735961ms] DMA CH B: ctrl: 0xC11F9007 count: 0x00000001
[735961ms] DMA CH C: ctrl: 0x0002900B count: 0x00000000
[735961ms] DMA CH D: ctrl: 0x00009809 count: 0x00000000
[735961ms] GPIO states: 0x157BFAFF
[735961ms] DBG ---- COMMAND: Unknown
[735961ms] DBG ------ OUT: 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 
[735965ms] DBG -- BUS_FREE
```

I've also seen the Amiga report block checksum errors occasionally.

## Cabling and Termination

I've wondered about cabling and termination issues, so have tested with a couple of different cables (one very short) without much change. I also thoroughly cleaned all the connector contacts. I've tried using an external active C50 terminator as well as the SCSIknife's internal termination.

## SCSI Modes and Speeds

At first I thought the problems might be associated with using high Synchronous speeds (perhaps with Reselection enabled), but the stability issues occur even with `MaxSyncSpeed=0` and `EnableSCSI2=0`.

Thanks for any help you can offer on troubleshooting further! I have a 50 MHz oscilloscope if that might help, but no logic analyser.

screwtop, 2025-09-02:
I should mention that I had problems when using the stock firmware that the SCSIknife came with (I don't think I have a copy of a log file from then, so I'm not certain which version that would have been). But the very first problem I had was more severe: an error dialog at boot time, presumably from the `1230scsi.device` driver, reporting "Unexpected phasechange". That seemed to be resolved by updating the firmware and/or cleaning the SCSI connectors.

screwtop, 2025-09-02:
I've also been experimenting with using the SD card in the microSD card slot in the SCSIknife, in case that makes a difference. At some point the card contents got somewhat corrupted (I don't think I ejected it during a write, but I can't be certain). The start of the block device then held the following:

```
00000000  22 22 22 22 22 22 22 22  22 22 22 22 22 22 22 22  |""""""""""""""""|
*
000001b0  22 22 22 22 22 22 22 22  22 22 22 22 22 22 22 a2  |""""""""""""""".|
000001c0  23 22 2e fe ff ff 22 22  22 22 22 26 b7 23 22 22  |#"...."""""&.#""|
000001d0  22 22 22 22 22 22 22 22  22 22 22 22 22 22 22 22  |""""""""""""""""|
*
000001f0  22 22 22 22 22 22 22 22  22 22 22 22 22 22 77 aa  |""""""""""""""w.|
00000200  22 22 22 22 22 22 22 22  22 22 22 22 22 22 22 22  |""""""""""""""""|
*
00400000  eb 22 b2 22 22 22 22 22  22 22 22 22 22 62 b2 26  |.".""""""""""b.&|
00400010  22 22 22 22 22 fa 22 22  3f 22 ff 22 22 22 22 22  |""""".""?"."""""|
00400020  22 26 b7 23 b7 3f 22 22  22 22 22 22 22 22 22 22  |"&.#.?""""""""""|
00400030  23 22 26 22 22 22 22 22  22 22 22 22 22 22 22 22  |#"&"""""""""""""|
00400040  a2 23 2b 36 3b 66 3a 6e  6f 22 6e 63 6f 67 22 22  |.#+6;f:no"ncog""|
00400050  22 22 66 63 76 33 32 22  22 22 22 22 22 22 22 22  |""fcv32"""""""""|
00400060  22 22 22 22 22 22 22 22  22 22 22 22 22 22 22 22  |""""""""""""""""|
*
004001f0  22 22 22 22 22 22 22 22  22 22 22 22 22 22 77 aa  |""""""""""""""w.|
00400200  72 72 63 63 22 22 22 22  22 22 22 22 22 22 22 22  |rrcc""""""""""""|
00400210  22 22 22 22 22 22 22 22  22 22 22 22 22 22 22 22  |""""""""""""""""|
*
004003e0  22 22 22 22 72 72 63 63  a3 e3 2f 22 72 3b 23 22  |""""rrcc../"r;#"|
004003f0  22 22 22 22 22 22 22 22  22 22 22 22 22 22 77 aa  |""""""""""""""w.|
00400400  22 22 22 22 22 22 22 22  22 22 22 22 22 22 22 22  |""""""""""""""""|
*
004005f0  22 22 22 22 22 22 22 22  22 22 22 22 22 22 77 aa  |""""""""""""""w.|
00400600  22 22 22 22 22 22 22 22  22 22 22 22 22 22 22 22  |""""""""""""""""|
*
00400c00  eb 22 b2 22 22 22 22 22  22 22 22 22 22 62 b2 26  |.".""""""""""b.&|
00400c10  22 22 22 22 22 fa 22 22  3f 22 ff 22 22 22 22 22  |""""".""?"."""""|
00400c20  22 26 b7 23 b7 3f 22 22  22 22 22 22 22 22 22 22  |"&.#.?""""""""""|
00400c30  23 22 26 22 22 22 22 22  22 22 22 22 22 22 22 22  |#"&"""""""""""""|
00400c40  a2 22 2b 36 3b 66 3a 6e  6f 22 6e 63 6f 67 22 22  |."+6;f:no"ncog""|
00400c50  22 22 66 63 76 33 32 22  22 22 22 22 22 22 22 22  |""fcv32"""""""""|
00400c60  22 22 22 22 22 22 22 22  22 22 22 22 22 22 22 22  |""""""""""""""""|
*
00400df0  22 22 22 22 22 22 22 22  22 22 22 22 22 22 77 aa  |""""""""""""""w.|
00400e00  72 72 63 63 22 22 22 22  22 22 22 22 22 22 22 22  |rrcc""""""""""""|
00400e10  22 22 22 22 22 22 22 22  22 22 22 22 22 22 22 22  |""""""""""""""""|
*
00400fe0  22 22 22 22 72 72 63 63  ff ff ff ff 22 22 22 22  |""""rrcc....""""|
00400ff0  22 22 22 22 22 22 22 22  22 22 22 22 22 22 77 aa  |""""""""""""""w.|
00401000  22 22 22 22 22 22 22 22  22 22 22 22 22 22 22 22  |""""""""""""""""|
*
004011f0  22 22 22 22 22 22 22 22  22 22 22 22 22 22 77 aa  |""""""""""""""w.|
00401200  22 22 22 22 22 22 22 22  22 22 22 22 22 22 22 22  |""""""""""""""""|
*
00492400  fa ff ff 2f ff ff ff 2f  ff ff ff 2f 26 22 22 22  |.../.../.../&"""|
00492410  27 22 22 22 26 22 22 22  27 22 22 22 2a 22 22 22  |'"""&"""'"""*"""|
00492420  2b 22 22 22 2a 22 22 22  2b 22 22 22 2e 22 22 22  |+"""*"""+"""."""|
00492430  2f 22 22 22 2e 22 22 22  2f 22 22 22 32 22 22 22  |/"""."""/"""2"""|
00492440  33 22 22 22 32 22 22 22  33 22 22 22 36 22 22 22  |3"""2"""3"""6"""|
00492450  37 22 22 22 36 22 22 22  37 22 22 22 3a 22 22 22  |7"""6"""7""":"""|
00492460  3b 22 22 22 3a 22 22 22  3b 22 22 22 3e 22 22 22  |;""":""";""">"""|
00492470  3f 22 22 22 3e 22 22 22  3f 22 22 22 22 22 22 22  |?""">"""?"""""""|
00492480  23 22 22 22 22 22 22 22  23 22 22 22 26 22 22 22  |#"""""""#"""&"""|
...
```

I've reformatted the drive and can continue testing. I can report that it's still showing I/O errors. FWIW, the SD card is brand new (bought last week).

screwtop, 2025-09-03:
Thanks for the suggestion - I've switched to v2025.08.19 and done a bit more testing. I also stripped the `zuluscsi.ini` down to:

```
[SCSI]
EnableSCSI2 = 0
MaxSyncSpeed = 0
```

I'm using [`devtest`](https://github.com/cdhooper/amiga_devtest) on the Amiga to do testing at the block level (to rule out any filesystem complications). Even with the limited settings, `devtest` occasionally reports read errors (e.g. "Read Fail 44 HFERR_SelTimeout at 3276800"). Here's the log from a test session where that happened:

```
[12ms] Platform: ZuluSCSI Pico
[12ms] FW Version: 25.08.19-release Aug 19 2025 17:09:14
[12ms] DIP switch settings: debug log 0, termination 1
[12ms] SCSI termination is enabled
[12ms] =========================================================================
[12ms]  Powered by Raspberry Pi
[12ms]             Raspberry Pi is a trademark of Raspberry Pi Ltd
[12ms] 
[12ms]  ZuluSCSI RPi platform support is developed by Rabbit Hole Computing
[12ms]  and provided to you under GNU General Public License version 3.
[12ms] =========================================================================
[13ms] Flash chip size: 2048 kB
[13ms] SCSI target/disk mode selected by DIP switch, acting as a SCSI disk
[29ms] SD card detected, FAT32 volume size: 30417 MB
[29ms] SD MID: 0x03, OID: 0x53 0x44
[29ms] SD Name: SD32G
[29ms] SD Date: 12/2024
[29ms] SD Serial: 0xC5F949F8
[30ms] Speed grade set to Custom reclocking system
[31ms] Using custom timings found in "zuluscsi_timings.ini" for reclocking
[59ms] Initial Clock set to 133000000Hz
[59ms] Reclocking the MCU to 133000000Hz
[59ms] Setting the SDIO clock to 22MHz
[59ms] After reclocking, system reports clock set to 133000000Hz
[59ms] ======== Reinitializing ZuluSCSI after reclock ========
[75ms] Reading configuration from zuluscsi.ini
[76ms] Active configuration:
[76ms] -- SelectionDelay = 255
[76ms] -- EnableUnitAttention = No
[76ms] -- EnableSCSI2 = Yes
[76ms] -- EnableSelLatch = No
[76ms] -- MapLunsToIDs = No
[76ms] -- EnableParity = Yes
[81ms] Finding images in directory /:
[82ms] -- Opening /HD4_Amiga.img for id:4 lun:0
[102ms] ---- Configuring as disk drive drive
[102ms] ---- Drive geometry from image size: SectorsPerTrack=48 HeadsPerCylinder=16 total sectors 196608 (divisible)
[102ms] ---- Read prefetch enabled: 8192 bytes
[102ms] -- Platform supports ROM drive up to 800 kB
[102ms] ---- ROM drive image not detected
[102ms] SCSI ID: 4, BlockSize: 512, Type: 0, Quirks: 0, Size: 98304kB
[112ms] Clock set to: 133000000Hz
[112ms] Initialization complete!
[180042ms] SD card removed, trying to reinit
[706326ms] SD card reinit succeeded
[706326ms] SD card detected, FAT32 volume size: 30417 MB
[706326ms] SD MID: 0x03, OID: 0x53 0x44
[706326ms] SD Name: SD32G
[706326ms] SD Date: 12/2024
[706326ms] SD Serial: 0xC5F949F8
[706326ms] Reading configuration from zuluscsi.ini
[706327ms] Active configuration:
[706327ms] -- SelectionDelay = 255
[706327ms] -- EnableUnitAttention = No
[706327ms] -- EnableSCSI2 = No
[706327ms] -- EnableSelLatch = No
[706327ms] -- MapLunsToIDs = No
[706327ms] -- EnableParity = Yes
[706334ms] Finding images in directory /:
[... 69 lines cut ...]
[26071ms] scsiDev.phase: 4
[26071ms] SCSI DMA state: WRITE
[26071ms] Current buffer: 0x00000200/0x00000200, next 0x00000000 bytes
[26071ms] SyncOffset: 0 SyncPeriod 0
[26071ms] PIO Parity SM: tx_fifo 4, rx_fifo 4, pc 31, instr 0x00004037
[26071ms] PIO Data SM: tx_fifo 8, rx_fifo 0, pc 27, instr 0x0000201A
[26071ms] PIO Sync SM: tx_fifo 0, rx_fifo 0, pc 0, instr 0x00000000
[26071ms] DMA CH A: ctrl: 0x01000011 count: 0x000001DC
[26071ms] DMA CH B: ctrl: 0x00020809 count: 0x00000000
[26071ms] DMA CH C: ctrl: 0x01008805 count: 0x00000001
[26071ms] DMA CH D: ctrl: 0x00009809 count: 0x00000000
[26071ms] GPIO states: 0x153DFC00
[26071ms] STACK 0x2003FE40:    0x1003BDAA 0x20000B65 0x20025684 0x00000001
[26071ms] STACK 0x2003FE50:    0x1003BDAA 0x00000000 0x143FFCFF 0x00000000
[26071ms] STACK 0x2003FE60:    0x20035BF0 0x0000000C 0x000003E8 0x2003A34A
[26071ms] STACK 0x2003FE70:    0xD0000128 0x00000000 0x00000000 0x20016915
[26071ms] STACK 0x2003FE80:    0xFFFFFFFF 0x2003A34A 0x20023E5C 0x00000000
[26071ms] STACK 0x2003FE90:    0x2001BBB0 0x00000000 0x00010160 0xFFFFFFF9
[26071ms] STACK 0x2003FEA0:    0xFFFFFFFF 0x00000000 0x20023E5C 0x20025368
[26071ms] STACK 0x2003FEB0:    0x20018CA9 0x10033B5B 0x20018CBA 0x81000000
```

During that session, the drive activity light remained on for some seconds at boot and the host system basically hangs.

One other observation: with the SCSIknife powered by the SCSI bus, the voltage at the +5 V pin on the board is only about 3.6 V. With it powered from the USB-C port, the voltage is more like 4.5 V (which still seems a little low to me).

screwtop, 2025-09-03:
I spoke too soon: NetBSD later had problems reading the ZuluSCSI disk using `dd`, and reports errors in the kernel messages:

```
bzivsc0: unexpected disconnect [state 5, intr 20, stat 81, phase(c 100, p 1)]; sending REQUEST SENSE
sd0(bzivsc0:0:4:0): bzivsc0: timed out [ecb 0x987f80 (flags 0x5, dleft 20, stat 0)], <state 2, nexus 0x987f80, phase(l 1, c 100, p 1), resid 7bdb, msg(q 0,o 0) >
sd0(bzivsc0:0:4:0): bzivsc0: timed out [ecb 0x987f80 (flags 0x45, dleft 20, stat 0)], <state 2, nexus 0x987f80, phase(l 1, c 100, p 1), resid 7bdb, msg(q 0,o 0) > AGAIN
```

Around that time ZuluSCSI reports the following:

```
[2630424ms] scsi_accel_rp2040_finishRead timeout
[2630424ms] SCSI DMA state: READ
[2630424ms] Current buffer: 0x00000006/0x00000006, next 0x00000000 bytes
[2630424ms] SyncOffset: 0 SyncPeriod 0
[2630424ms] PIO Parity SM: tx_fifo 0, rx_fifo 0, pc 6, instr 0x000060C8
[2630424ms] PIO Data SM: tx_fifo 0, rx_fifo 4, pc 19, instr 0x00005056
[2630424ms] PIO Sync SM: tx_fifo 0, rx_fifo 0, pc 0, instr 0x00000000
[2630424ms] DMA CH A: ctrl: 0x01020021 count: 0x00000006
[2630424ms] DMA CH B: ctrl: 0xC11F9007 count: 0x00000001
[2630424ms] DMA CH C: ctrl: 0x0002900B count: 0x00000000
[2630424ms] DMA CH D: ctrl: 0x00009809 count: 0x00000000
[2630424ms] GPIO states: 0x157BFEFF
[2635428ms] scsi_accel_rp2040_finishRead timeout
[2635428ms] SCSI DMA state: READ
[2635428ms] Current buffer: 0x00000006/0x00000006, next 0x00000000 bytes
[2635428ms] SyncOffset: 0 SyncPeriod 0
[2635428ms] PIO Parity SM: tx_fifo 0, rx_fifo 0, pc 6, instr 0x000060C8
[2635428ms] PIO Data SM: tx_fifo 0, rx_fifo 4, pc 19, instr 0x00005056
[2635428ms] PIO Sync SM: tx_fifo 0, rx_fifo 0, pc 0, instr 0x00000000
[2635428ms] DMA CH A: ctrl: 0x01020021 count: 0x00000006
[2635428ms] DMA CH B: ctrl: 0xC11F9007 count: 0x00000001
[2635428ms] DMA CH C: ctrl: 0x0002900B count: 0x00000000
[2635428ms] DMA CH D: ctrl: 0x00009809 count: 0x00000000
[2635428ms] GPIO states: 0x157BFAFF
[2640432ms] scsi_accel_rp2040_finishRead timeout
[2640432ms] SCSI DMA state: READ
[2640432ms] Current buffer: 0x00000006/0x00000006, next 0x00000000 bytes
[2640432ms] SyncOffset: 0 SyncPeriod 0
[2640432ms] PIO Parity SM: tx_fifo 0, rx_fifo 0, pc 6, instr 0x000060C8
[2640432ms] PIO Data SM: tx_fifo 0, rx_fifo 4, pc 19, instr 0x00005056
[2640432ms] PIO Sync SM: tx_fifo 0, rx_fifo 0, pc 0, instr 0x00000000
[2640432ms] DMA CH A: ctrl: 0x01020021 count: 0x00000006
[2640432ms] DMA CH B: ctrl: 0xC11F9007 count: 0x00000001
[2640432ms] DMA CH C: ctrl: 0x0002900B count: 0x00000000
[2640432ms] DMA CH D: ctrl: 0x00009809 count: 0x00000000
[2640432ms] GPIO states: 0x157BFAFF
```
At this point I'm thinking I should try another SD card, and also test the SCSIknife on a different SCSI host system altogether. I can probably set up a test system with the SCSIknife installed internally on a 50-pin ribbon cable...

screwtop, 2025-09-07:
I tested the SCSIknife with the SanDisk card in a Linux PC with an Adaptec 39160 card, using the internal ribbon cable. The only things in common with the Amiga setup are the SCSIknife and the SD card. The machine did hang unusually (like, completely frozen) a couple of times during initial testing, but following that, SCSIknife worked without any apparent problems on that machine, reliably achieving 9.5 MB/s reads. The SCSI card hadn't been used in a while and I wondered if tarnished pins or ageing capacitors might have been the reason for the initial hangs.

I then tested with the Amiga again using a different drive enclosure, but it still showed the same kinds of problems. However, I swapped the SanDisk microSD card for an old Apacer 4 GB Class 6 full-size card (mentioned earlier), and that now seems to be working without problems. I ran `devtest` for about 8 hours overnight with no errors reported. It's a bit strange that the Apacer card works fine, because I was using that card during the very first testing that I did, and bought the SanDisk after having issues. I note that the Apacer card does limit the transfer rates somewhat compared to the SanDisk, with a max of ~6 MB/s read. This successful test on the Amiga was using just the SCSI bus power, and using the SCSIknife's internal termination.

I'm not sure if/how this helps, but I can experiment with some different SD cards and see if a clearer pattern emerges.

aperezbios, 2025-09-07:
@screwtop I'd start by measuring the input voltage on pin 25 of hte SCSIKnife. It should be +5VDC, and If it's too low, the voltage drop once it goes through a diode or two may be low enough that the regulated voltage to the SD card is simply too low for it to operate properly.  The RP2040 itself can run at voltages way below that of most SD cards, which can obviously cause a breakdown of communication.  Since the SCSIKnife does not include any on-board bulk capacitance (electrolytic or tantalum capacitors, which all actual ZuluSCSI boards include for reasons such as this), SCSIKnife is inherently far less able to manage this sort of situation, electrilcally speaking, unless you separately supply the SK with external power via USB.

I'm going to convert this issue into a GitHub discussion, since it's non-actionable for us on the firmware side.